Carvera-Community/Carvera_Community_Firmware #45: Track the clamp/unclamp status of the ATC Collet

SergeBakharev, 2024-11-15:
Currently the clamp status is not tracked by the machine. There is no sensor for “clamped” only unclamped 
since depending on the bit the actual clamping distance might be different

> M490 homes the atc
> M490.1 just clamps
> M490.2 homes and unclamps
> M6T# drops the tool, homes the atc while just above the next tool, then grabs the tool and clamps

It might be a cool improvement to the firmware to track it. Tool changes would be slightly quicker, can prevent the spindle from spinning when it's unclamped preventing a halt condition, and from the controller could report on the status.

faecorrigan, 2024-12-06:
the status is tracked, just not exposed. in the next release I have a function that can grab the status from anywhere. How would we like to report it to the controller?
@SergeBakharev

SergeBakharev, 2024-12-06:
Are there any existing patterns we should follow? For example is there a way to check if the machine is in the laser mode? 

It would be good to have it included in the diag command response, but not critical.

faecorrigan, 2024-12-07:
we should look at the output of M114,M115 and M119 to see how they operate. 

otherwise the diagnose and update strings could get appended, but I dont know if we want to deal with the ramifications there.