Carvera-Community/Carvera_Community_Firmware #72: delta_steps_per_mm missing?

L0rdS474n, 2025-05-04:
Switched to Community firmware and keep getting this error/warning on my Carvera Air

[image]

Is that something I need to do by myself?

faecorrigan, 2025-05-07:
it should default to the correct value even if it isn't in the config, I am also working on an update that should fix this

L0rdS474n, 2025-05-28:
How would I know this "4th axis harmonic drive", I have the 4th axis that I got with the Carvera Air.

WARIO2412, 2025-05-29:
The 4th axis that ships with the Air is always the harmonic drive version. So in your case, you have to set the steps to the value I posted before. Otherwise the 4th axis would barely move.

L0rdS474n, 2025-05-29:
Thanks! I don’t use it that often, so that would’ve been quite the surprise!

WARIO2412, 2025-05-29:
Same happened to me :)