Carvera-Community/Carvera_Community_Firmware #77: Tool probe feedrate too high during Z calibration 3 axis cnc probe – causes 3D probe to hang in triggered state

Joqiz, 2025-05-22:
Firmware: Carvera.Community.Firmware.V1.0.3c1.0.5.bin
Machine: Carvera Air
Controller: Community controller (tested on Android & Windows)

Hi and thank you for all the great work on this firmware and controller!

I'm using a standard 3D touch probe (normally open), as shown in this guide:
https://www.instructables.com/Carvera-Touch-Probe-Modifications/

According to the probe's specifications and common practice, the maximum safe probing feedrate is 50 mm/min — especially during downward movement (Z probing).

Problem:
This issue is about the tool calibration function — the button in the community controller UI that automatically calibrates tool length (Z) using the tool setter block (probably via M500/M501).

Even when using "slow" probing, the feedrate during tool calibration appears to be too fast for sensitive 3D probes.

What happens:
The probe LED stays red after calibration (triggered state)
The probe returns to normal (green) only after slight wiggling
This suggests internal overstress of the probe mechanism
Long term, this could lead to wear, false triggers or calibration errors

Suggested improvements:
Allow the tool probing feedrate to be configurable, e.g.:
config-set sd tool_probe.feedrate 50
-Apply slow feedrate only to tool 0 (3D probe), and not to normal cutting tools

Let me know if I can help test any changes — thanks again for building this amazing tool!

J

SergeBakharev, 2025-05-23:
Some more detail and discussion is present in a Controller issue here: https://github.com/Carvera-Community/Carvera_Controller/issues/197

WARIO2412, 2025-05-25:
It has been discussed in the discord, that the default slow probe speed is 60 mm/min and therefore not far from the recommended 50 mm/min. It seems like the probe itself is faulty here. The fast probing speed still wouldn't hurt the probe since the distance the tip can travel before hitting some mechanical boundaries is far greater than the brake distance of the spindle while probing.

Unless you add something to this the next days, I will close this :)

closed 2025-05-26 by WARIO2412