Carvera-Community/Carvera_Community_Firmware #78: Carvera Air - 3D Probing does not work

GeorgeM102, 2025-05-22:
Hello 

3d probing does not seem to work on Carvera air. The  probing sequence does not initiate at all on carvera air.
Also there is no power to the "probe laser" unless you go on diagnostics and power it manually. If the power is on i can see the signal going green when pushing the probe. When calibrating the 3d probe i get a dead probe error.

Thanks

Joqiz, 2025-05-22:
Same here, but for the moment I wired it via USB and it works perfect. I would like it to be exactly as described at https://www.instructables.com/Carvera-Touch-Probe-Modifications/ for the Carvera Air. It should be an easy fix

WARIO2412, 2025-05-25:
> Hello
>
> 3d probing does not seem to work on Carvera air. The  probing sequence does not initiate at all on carvera air.
> Also there is no power to the "probe laser" unless you go on diagnostics and [...]

How did you flash the community firmware?
The probe dead part can either be, that you are still on the stock firmware or the wiring is bad.

WARIO2412, 2025-05-29:
You can also use Tool 0 for 3 axis probing, if you enter the following in the mdi:

config-set sd zrobe.tool_zero_is_3axis true

For me I don't use the stock wired probe anymore and therefore I allow tool 0 to be the 3axis probe.

GeorgeM102, 2025-05-29:
Ok so I now checked firmware is 1.0.5 probe TP06 . I've set the probe to zprobe as per above and when i got to calibrate  i get alarm :ALARM: Probe failed to trigger within safety margin (0.10mm)

WARIO2412, 2025-05-29:
Issue has been fixed by flashing the correct firmware. The calibration safety margin had to be set to 0.2 in order to be able to calibrate the probe. After that it worked with a margin 0.1 aswell. Unclear to why that happened.
Command is: "config-set sd zprobe.calibration_safety_margin 0.1"