Carvera-Community/Carvera_Community_Firmware #87: Bug: Carvera Air 4th axis not working on latest firmware

aaronmgd, 2025-06-10:
Bug: Carvera Air 4th axis not working on latest firmware. A axis does not register. It does home when hitting home but does not register any other commands.

WARIO2412, 2025-06-10:
Can you give more detail? Community Controller or stock?
Does it home fast or really slow? What commands are you handing over?
What is the output of "config-get sd delta_steps_per_mm"?

WARIO2412, 2025-06-14:
I'm sorry, put this in the mdi:

"config-set sd delta_steps_per_mm 888.888889"

WARIO2412, 2025-06-14:
https://github.com/Carvera-Community/Carvera_Controller/issues/233

aaronmgd, 2025-06-15:
> 888.888889

should i set this in setting as well? i entered that into midi but in settings it did not change

aaronmgd, 2025-06-15:
> It should change after a restart. But it would be best to use the latest community controller 0. 8.2 and then change it in the settings there.

this did not solve issue. during job A axis does not rotate.

WARIO2412, 2025-06-26:
Fixed by erasing the eeprom after the upgrade to the latest community firmware

closed 2025-06-26 by WARIO2412