Carvera-Community/Carvera_Community_Firmware #103: Rotary axis not turning in run processing

sheimend, 2025-07-23:
My rotary axis did not turn during the processing of a simple 4-axis milling job. The rotary axis is installed and connected, and it works when I move it manually through the Carvera Controller. Prior to this community firmware, it worked as expected. 

When the milling job runs, X-Y-Z moves are executed correctly. But it's almost as if the A-axis is being ignored. It doesn't rotate, and the controller GUI always shows it in 0-position. 

I'm on firmware version Carvera.Community.Firmware.V1.0.3c1.0.6

I've attached the CNC job and my Fusion 360 file containing the CAM program. 

[Simple 4-axis.zip](https://github.com/user-attachments/files/21394337/Simple.4-axis.zip)

sheimend, 2025-07-23:
Just tried to repro on the 1.0.8c firmware with the 25.7.23 controller. The A-axis now turns, but it only rotates a few degrees total, instead of making the several complete revolutions expected in the NC.

sheimend, 2025-07-23:
That did the trick! Thanks so much.

WARIO2412, 2025-07-23:
You're welcome!