Carvera-Community/Carvera_Community_Firmware #109: Thread milling moves often distort in Z-axis

sheimend, 2025-07-29:
When milling threads (internal or external) I'm finding that the movement of the machine isn't matching the intent of the gcode. 

For instance, in the attached external thread, the machine makes 2 circles, but does not drop down in Z. So I end up with a single groove, not a milled thread. In the attached internal thread, the machine seemed to stretch the Z axis and create one groove with a pitch that was way too high. In both cases, they obey the X-Y coordinates correctly. 

[thread moves.zip](https://github.com/user-attachments/files/21498368/thread.moves.zip)

[image]

I'm creating the thread operation in Fusion 360 with the 2D thread feature. The gcode looks correct in Fusion's simulation, and in the Carvera Controller GUI. 

If I reboot the machine or do other voodoo like re-uploading the file, it will sometimes mill the correct thread. (Sorry, the exact repro is a little inconsistent). 

I'm on FW build Carvera.Community.Firmware.V1.0.8c_beta and controller build v25.7.23.

sheimend, 2025-07-30:
I spoke too soon.... 

I ran the attached NC program:
[Threaded v4 M face 1018 v1.zip](https://github.com/user-attachments/files/21518991/Threaded.v4.M.face.1018.v1.zip)

The thread feature starts at line 5566. I intentionally used a tool that wouldn't make contact so I could record the motion path in time lapse and see if it looked correct: 
https://photos.app.goo.gl/aZqiReK7s9iewqrN8

The trial did indeed look correct. So I paused the program, executed "goto 5565" and ran that section again with my thread mill. When it ran the second time, this is what I got. Just a circle with no Z change. 

[image]

So, I restarted the program, jumped to 5565, and ran it again and this time it milled the thread correctly. 

I don't know why this bug is only a sometimes repro, but I hope this information helps. 

I'm now on controller version v.0.10.0 and V1.0.9c firmware for the you see in this repro.

WARIO2412, 2025-08-08:
Can you update to 1.0.10c? Not that we found a bug regarding this directly, but there was a bug that wasn't obvious and also didn't always appear. So let's try this first if you haven't so far.