Carvera-Community/Carvera_Community_Firmware #123: Carvera Air Soft Limit Triggered by File

jterblanche, 2025-08-12:
Peculiar situation on the Carvera Air. If I run the following commands manually line by line, all works perfectly, but if I put it in a file and run the file, the Carvera Air locks up with a "Soft Limit Triggered":

```
(Program Name: 1001)

(Tool Information:)
(T999990  3D Probe      D=2.0 CR=1.0 - ZMIN=10.0 - probe)

G90 G94
G17
G21

(Start of operation: probing-xy-outer-corner 2)
M6 T999990
G59
G0 X-2.0 Y32.0
Z52.0
(Begin probing-xy-outer-corner)
(WCS used during probing: G59)
(WCS that will be updated: G54)
G1 X-2.0 Y32.0 Z32.0 F1000.0
G38.3 Z-22.0 F1000.0
G38.3 Y-12.0 F1000.0
G38.2 X12.0 F204.0
G91 G1 X-2.0 F1000.0
G38.2 X4.0 F102.0
G10 L20 P1 X-1.0
G1 X-4.0 F1000.0
G1 Y12.0 F1000.0
G38.3 X12.0 F1000.0
G38.2 Y-12.0 F204.0
G1 Y2.0 F1000.0
G38.2 Y-4.0 F102.0
G10 L20 P1 Y1.0
G1 Y4.0 F1000.0
G1 X-12.0 F1000.0
G90 G0 Z52.0
(End probing-xy-outer-corner)
X-2.0 Y32.0 Z52.0
M5
M852
G28
M30
```

WARIO2412, 2025-08-12:
Can you show the Mdi output and on what line it reaches the soft limit? And maybe a picture of the setup

jterblanche, 2025-08-17:
What's the fix for this then?

WARIO2412, 2025-08-17:
We will need to update the community controller. The coordinates are being generated in the controller, not the firmware. I'll open an issue there.