Carvera-Community/Carvera_Community_Firmware #128: Random Shutdowns on Latest Release (V1.0.11c)

cmess0822, 2025-08-15:
## Description
While running G-code generated from Fusion 360 using the community Carvera post and machine files, the machine stops mid-job, becomes unresponsive, the enclosure light turns off, and requires a full power cycle to recover.  

This occurs more quickly when smoothing is enabled in Fusion 360 (producing more arc G-code commands), but still happens without smoothing after a longer run time.  

I do not have conclusive evidence, but the behavior may be consistent with a memory management issue.

---

## Steps to Reproduce
1. Use Fusion 360 CAM with the community Carvera post and machine files.
2. Create an adaptive clearing toolpath.
3. Enable *smoothing* (optional, generates more arc g-codes, but triggers the issue sooner).
4. Run the generated G-code on the Carvera (C1).

---

## Observed Behavior
- Cleaning and securing connections and replacing the SD card did not resolve the problem.

---

## Additional Notes
- Downgrading to the latest official Makera firmware eliminates the issue.
- Previous community firmware did not seem to exhibit the problem.
- Thorough cleaning and hardware checks did not help.
- New SD card installed — no change.

---

## Environment
- **Machine:** Carvera C1 (Kickstarter)
- **Firmware:** Latest community firmware (issue present)
- **Official Makera Firmware:** Latest release (no issue)
- **CAM Software:** Fusion 360
- **Post Processor:** Community Carvera post & machine files

cmess0822, 2025-08-16:
I do not have the original g-code I was using as I was experimenting with different settings at the time to get the part made.  However, I generated an equivalent program to see if I could reproduce the issue.

I used Firmware V1.0.11c for this test.  Probed Z a few inches off the bed to do a dry run with no actual milling.
This g-code does reproduce the issue on my hardware.  Crashes at about the 20min mark after the run command is sent and the controller indicates it stopped at line 4083.  The enclosure light goes off and movement stops, however this time it reset on its own.  During my previous tests I had to manually power cycle the machine to bring it up again.

I zipped the g-code since github does not seem to accept .cnc files as attachments.
[Issue128_Sample_Gcode.zip](https://github.com/user-attachments/files/21812609/Issue128_Sample_Gcode.zip)

WARIO2412, 2025-08-16:
Did you have the bed level turned on?

WARIO2412, 2025-08-16:
You are on the C1, note the AIR?

WARIO2412, 2025-08-16:
Nice I was able to reproduce with the file. Will test with the new firmware now

saamirt, 2025-09-17:
Also running into random shutdowns on 1.0.11c on the Carvera Air (model = CA1, 2, 1, 5693)